ClemensElflein/OpenMower #1: OpenMower is not Open Source

abetusk, 2022-04-07:
The common definition of open source is that the source code and other digital artifacts can be used for commercial purposes. The non-commercial licensing makes this project not open source. A list of resources discussing the definition of "Open Source" is given below. The Open Source Initiative (OSI) has a [list of approved licenses](https://opensource.org/licenses/category). The Creative Commons licenses that are libre/free/open are [CC0](https://creativecommons.org/share-your-work/public-domain/cc0/), [CC-BY](https://creativecommons.org/licenses/by/4.0/) and [CC-BY-SA](https://creativecommons.org/licenses/by-sa/4.0/).

Please consider removing the non-commercial clause in the license to make this project open source.

If you decide to keep the non-commercial licensing, please consider renaming the project to something other than "OpenMower" as the term "Open" holds connotations of it being open source and causes confusion. Additionally, if you decide to keep the non-commercial license, please remove the "open source" project goal in the `README.md` file.

For projects that aren't libre/free/open source but still wish to make their source available, a good compromise I've found is to indicate that the project has "source available" rather than specifying it is "open source".

---

[Open Source Initiative](https://opensource.org/osd)

> The license must not restrict anyone from making use of the program in a specific field of endeavor. For example, it may not restrict the program from being used in a business, or from being used for genetic research.

[Wikipedia's entry on Open Source](https://en.wikipedia.org/wiki/Open_source)

> Licenses which only permit non-commercial redistribution or modification of the source code for personal use only are generally not considered as open-source licenses.

ClemensElflein, 2022-04-07:
Thank you for bringing this to my attention. I was not aware that commercial use was one of the requirements for "open source" projects. I'll have to think about where to go with this project. Sorry for the confusion.

pbs-websuntangled, 2022-04-11:
The author has put a huge amount of effort into this 

And it's good to name things properly so communication is clear 

And I'm so happy that someone has thrown themselves at this problem 

Maybe there are some collaborators that could create a good business out of this with the author?

ClemensElflein, 2022-04-11:
Thank you for your comment @pbs-websuntangled. I don't really understand the instant thumbs down to be honest.
Instead of making it closed source, I chose to share the progress for free with the only exception that you don't just go manufacture and sell it before I have the chance to do so myself.

I think that settles it for me. Non commercial use it is for now. I'll remove "open source" from the goals as you requested, but I'll keep the name since I don't think that your association between "open" and "it has to be free for commercial use" is enough to justify that.

The project is open in the sense that anyone can check the sources and replicate this without paying me anything. Designs, code and instructions will be free.

closed 2022-04-11 by ClemensElflein

Hypfer, 2022-04-11:
> I don't really understand the instant thumbs down to be honest.

You're abusing the open source community to push your own personal goals for your own personal benefit while pretending that it's all altruistic. It isn't. It's just marketing.

Stop doing that.
We don't need "entrepreneurs" pretending to be foss maintainers and ruining the words "Open Source" in the process.

hearnia2k, 2022-04-11:
I agree with the creator.  

I would never consider that something being open source has any relation to whether it can be used commercially.  Open source means anyone can go and download and look through the code, and even compile it if they want to; it's openly available.

FadeFx, 2022-04-11:
@ClemensElflein cgeck this: http://www.gnu.org/philosophy/selling-exceptions.en.html

i guess this is what you would actually want

abetusk, 2022-04-11:
@ClemensElflein @hearnia2k regardless of what your own personal definition of "open source" is or what using "open" in the name of a project is, there is a widely held community definition of what those terms signify, which, at least for the "open source" term, I've linked to in the initial comment of this issue. Regardless of what you think it should mean or what you take it to mean, people at large will take it to mean that it is libre/free/open, where libre/free/open specifically respects users freedom which includes commercial reuse.

Under generous assumptions, using "open" in the name creates confusion among people wanting to use this  project but then are restricted from doing so because of the non-commercial license. Any use of the term "open source" when there is a non-commercial license is falsely representing what the licensing terms are.

There is a gradation between closed source, source available and free/libre/open source. Closed source is source that is not available to the public at large and is protected under copyright. Source available is source that is available for review but has copyright protections in place so that it can't be used or copied without the copyright holders express permission. Libre/free/open source is source that is available and has licensing terms that allow for distribution, alteration and use, even for commercial purposes.

This project, as the author has communicated it in this issue, is a "source available" project.

@FadeFx what you've described is dual licensing, which is allowing companies to re-license the code under a standard license so they don't need to worry about any licensing terms that might not compose well with other parts of their software stack. As far as I understand the intent of the author in this thread, dual licensing is **not** applicable to this case as they want to make sure commercial re-use, even under a libre/free/open license, is expressly forbidden without explicit approval.

FadeFx, 2022-04-11:
@abetusk you are right about that this is actually not what @ClemensElflein intended, but as Richard Stallman stated selling exceptions is also not dual licencing actually. 
@ClemensElflein so you intend to use this project commercially at some point later and want to keep others from doing so with your work? Then this should really not be open source. Probably Creative Commons has something to cover your needs, but i dint know how well that goes with code.
However, you could probably profit from others using this in their projects and getting back improvements they did as well if you release it under the right licence, by requireing open source licencing from them as well. This might create a win win situation finally.

hearnia2k, 2022-04-11:
@abetusk, I disagree.

For example, what would the point of the F in FOSS be if open source automatically means it's commercially usable for free?  

As for the naming, making assumptions based on the name OpenMower seems like your own downfall to me.  It's open in that the code is openly available. 

Your own definitions of closed, available and free don't line up well, either.  Closed source is pretty clear.  However, free has nothing to do with source availability, for example, and you have seemingly grouped this with libre and open.  There are plenty of programs that are free that are either open or closed source.  Plus, your 'source available' option doesn't allow for any community input, modification, redistribution, or implementation - I think these would be allowed here; but so long as it's not part of a commercial offering.  So, to me, none of the terms fit, as you describe.

Wikipedia starts by saying "Open source is source code that is made freely available for possible modification and redistribution".  Your quote from Wikipedia even qualifies it's statement by saying 'generally', ie, there are licenses that do not allow commercial use, while the software is still open source.  

Open source is a valid set of words to describe any project where the code is openly available.  Some people may expect or hope for more, or certain other conditions, but the term is not reserved or owned in exclusivity to those people, it's just a set of words.

abetusk, 2022-04-11:
@FadeFx the author has removed the term "open source" from the relevant documents (the `README.md` file). While I think having "open" in the name of the project is misleading, this is not outright claiming it's "open source" so I'm not going to push this point further.

The author has put everything under a CC-BY-NC-SA license (Creative Commons, By Attribution, Non-commercial, Share Alike) which is a "Creative Commons" license but is not a libre/free because of the non-commercial aspect of it.

Again, to be clear, the author is not directly claiming the project is "open source" so this issue (#1) is resolved.

hearnia2k, 2022-04-11:
Thanks for a good response.  

I don't have specific links / resouces to show my view, however, I would say that both Wikipedia and your own explanations are using qualifiers, which allow for my view point as well; and that in itself makes the position of calling a not commercially usable project open source to be valid, even if your viewpoint is accepted.

I see the point about free being freedom vs price, while my earlier comment was more focussed on the price aspepct.  I realize you had included it with libre too, but I understood the slash to be more like alternatives rather than synonyms; which reinforced that the libre and free were indepedent items, rather than the same item expressed in a different way.  

To me a project being open source is more about users being able to read, and contribute to the project; it's more that the project is a community thing, rather than any kind of commercial aspects, and I would not expect a community project to be able to be used commercially without special licensing.  

However, it does appear my view against the more common viewpoint, but it does make the term 'open source' much more muddy by trying to add the extra element of commercialization to two normal words, especially when doing so creates a gap for how to describe a project that open source, but not for commercial use. 'Source available' suggests the project is not looking for contributions, and is simply showing their work for example.  

Anyway, as you pointed out the creator removed open source from the readme, which seems like a reasonable compromise without changing name.

EmanuelLoos, 2026-01-07:
@ClemensElflein

Perhaps you would be willing to make the firmware truly Open Source while the hardware stays source-available for non-commercial use only? Open Source firmware could be a good selling point for your hardware while it probably would not allow any greedy company to take advantage of your work without paying for it, especially if you decided to use a copyleft license like the GNU GPLv3 for it. After all, who is going to pay for a software that can be downloaded for free from the internet anyway? And even if some company where to take advantage of your software, they would have to let you take advantage of their modifications as well, so you would profit from it.

PS: I am interested in buying the V2 Hardware from you for a JohnDeere mower and would be even more interested if the firmware was truly open source.

I'd like to avoid the Discord platform. You can contact me via email: mail@emanuel-loos.eu