ClemensElflein/OpenMower #2: Website SSL Error

admbradford, 2022-04-07:
Receiving an `ERR_SSL_PROTOCOL_ERROR` when visiting your blog site https://x-tech.online/

ClemensElflein, 2022-04-07:
Thank you for the report. It's working fine for me even in a private tab. Which browser are you using? What's the error message you're getting?

admbradford, 2022-04-08:
I’m on mobile chrome v100.0.4896.77 iOS 15.4.1

here’s a screen shot

[image]

admbradford, 2022-04-08:
I checked off my home network and it worked fine, I have some aggressive protection from my router. It’s likely due to the second cert

[SSL Server Test: x-tech.online (Powered by Qualys….pdf](https://github.com/ClemensElflein/OpenMower/files/8451909/SSL.Server.Test.x-tech.online.Powered.by.Qualys.pdf)

ClemensElflein, 2022-04-08:
Thank you for the info! I'm looking into it. To be honest, I'm just using the Let's Encrypt Certbot and was glad that it seemed to work out. There has to be some strange configuration still.

closed 2022-09-22 by admbradford