ClemensElflein/OpenMower #3: Option to mow in a randomized pattern?

jh83, 2022-04-08:
Is it possible to have the mower to cut the grass in a randomized pattern  and only use the perimeter as a "limit" for which it should stay within?

Having the mower cut on the same lines over and over again will create track's in the lawn. There is a reason to why husqvarna has spent a lot of time developing how the robot should find its way back to the charger, where it actually varies its distance to the perimeter cable to avoid creating track's.

Husqvarna has now released a gps rtk version of its automower (which is very expensive) and I think that stiga will release one soon as well.  As far as I know, this still cut in a randomized pattern, while using gps for the perimeter limits.

Very good project though!

R5fan, 2022-04-11:
If you want to avoid making "tracks" or stripes (which many people try to make on purpose btw), the much better solution is not reverting to random navigation, but randomizing either the starting point (so tracks are offset a bit on each session, although with the narrow mowing width of most robots that may not yield much) or rotate the orientation of the tracks by a random angle.

jh83, 2022-04-11:
A robotic lawn mower does not create the "stripes" which many People (including me) like even if it would cut in a straight pattern.

I agree on the inefficiency in the randomized pattern. But there is a reason to why the GPS mowers are sticking to the randomized (which not really is randomized) pattern on their rtk-gps models.

A robotic lawn mower, when used as intended, creates a much nicer lawn compared to a normal cutter. The reason behind it is that the robotic "always" cuts and does it from different angles. Moss and other none-grass things does not like it, and is replaced by the grass. - So there are different views on what a robotic cutter is, if it is intended to cut the lawn as quickly as possible, or if it is intended to "take care" of the lawn.

If running on straight lines, then care needs yo be taken to the uncut grass that remains when the wheel presses down rhe grass, and then is unable to cut that grass on the next line and so on.

Here is an interesting article on the line vs random cutting. It says that there is only a 30% time difference between the two. https://robolever.com/random-vs-systematic-robotic-mowers-which-is-better/

As said. We like things differently, so If it would be possible to run either straight lines, or in a random pattern and use gps for perimeter then it would be great

pbs-websuntangled, 2022-04-11:
Agreed 👍

R5fan, 2022-04-11:
Mowing from random angles matters, but systematic mowing with a randomly rotated pattern every session would yield the same result. I dont see the downside. Heck, since his code uses slic3r code, you can even use any other infill pattern you want, plenty to chose from, all of them far more efficient than random.

Speaking of efficiency, if you have a square lawn with no obstacles, _maybe_ the 30% less efficient claim is right, but for sure its not in my garden, where my robot can mow for 3-4-5 days and still not cover every patch. If it mowed systematically, knowing its width and estimating speed, I guess it would take ~3-4 hours?  Even if its 24hr, that is a massive difference. I used to have a random navigation robot vacuum, and that is okay, because you cant tell if it "missed" a spot for a few days. Lawn mower is different, if you cant mow for a week due to rainfall, and then it takes almost a week before the robot to "catch up" and get everywhere. So during that  week you will have patches of freshly mowed grass next to nearly 2 week growth, not very pretty. More than once I took my old mower and mowed my lawn by hand to avoid exactly that. 

Anyway, if that is what you want, who am I to stop you, but it makes no sense to me.

pbs-websuntangled, 2022-04-11:
For me the point is not so much about wanting randomness 

Rather it's good to cut the grass from different angles with different patterns

It's definitely true that for real world lawns, it can be 4 or 5 days of permanent mowing before the same hit is hit twice 

Structured mowing with  rotated efficient row by row mowing would be a great feature

This project is amazing 👍👍👍

R5fan, 2022-04-11:
> Structured mowing with rotated efficient row by row mowing would be a great feature

Agreed! Because random navigation doesnt result in even distribution. To illustrate, here is my back lawn:
[image]
The visible tracks give a hint (I would really LOVE to create a coverage map somehow from my timelapses, or even be able to simulate random navigation on my lawn and see the result, im just curious what the coverage looks like, and I know its not evenly distributed);  the mower will cross the center of the image dozens of times before ever reaching the left corner. I also really do not like the "look" of those tracks. They fade, but if you mow every day, they are there every day especially if its a bit humid.

kevinmce, 2023-06-30:
There is a mow pattern orientation in the mower config, this should be relatively easy to have randomized at the start of each mow should the user input that option.

closed 2026-09-03 by ClemensElflein