ClemensElflein/OpenMower #4: How to start ti build the more?

chicco83, 2022-04-10:
Hi.
I'm very interested in the project but It Is not clear how to start the hardware part. Which parts to buy, if It Is mandatory to have a dedicated PCB produced ecc

r00li, 2022-04-13:
Yeah I too would be interested in a bit more detailed guide of what exactly needs to be purchased, which PCBs made, ... 

I had a look at the BOM and it seems a bit incomplete - it seems to only cover the main board components, but doesn't include things like the GPS receiver, ...

GruNyv, 2022-04-16:
Based on the video he posted, the gps receiver is a simplertk2b (with a zed-f9p GNSS module) and a u-blox multiband antenna. These are relatively cheap, around 200 euros, considering that other brands usually cost several thousand euros.

Kisses99, 2022-04-17:
Maybe it's even cheaper to hook up an old smartphone.

GruNyv, 2022-04-17:
@Kisses99 The problem is that your smartphone is not RTK compatible. You need a GNSS module that supports RTK, such as the ZED-F9P.

Kisses99, 2022-04-17:
Ah got it.

GruNyv, 2022-04-17:
Yes, the cost of RTK certainly is a barrier to entry. This was especially true before the ZED-F9P, where it was mostly reserved for the industry and extreme hobbyists. Another issue is that you might need 2 of them, where one acts as a base station, if there are no freely available base stations near you through i.e. RTK2GO and you don't want to pay for a commerical subscription

FadeFx, 2022-04-17:
@Kisses99 there are cheaper optiins as well, but i guess they dont provide enough precission for a task like that. Did you ever check the precission of your smartphone gps? its most likely not below 5m...

GruNyv, 2022-04-17:
Unless you use some sort of differential gnss (which RTK is a subgroup of), you can't expect better accuracy than 3-10 meters depending on the conditions

Kisses99, 2022-04-17:
@FadeFx right that's what I just figured.  However I'm still concerned that gps accuracy may be significantly impacted by shades of trees.  For an open yard this may just be fine.

GruNyv, 2022-04-17:
Yes, the tree shades and building is what I am concerned with. I have been working with this exact model during this spring and it often has troubles when you enter non-open areas

FadeFx, 2022-04-17:
could it be possible to compensate this by also taking in account your actual movement?

GruNyv, 2022-04-17:
You certainly can predict your movement using multiple sensors using a kalman filter. However, it might work to simply drive randomly when you are in a region without signal, since this is what they already do. The issue is of course the possible lack of boundary wire

mischla, 2022-04-17:
> Yes, the cost of RTK certainly is a barrier to entry. This was especially true before the ZED-F9P, where it was mostly reserved for the industry and extreme hobbyists. Another issue is that you [...]

Is the German SAPOS system which provides Ntrip in RTCM 3.1 format what would be needed here?

GruNyv, 2022-04-17:
@mischla Most network RTK providers which uses RTCM 3 should work, but you should check out which satellite configurations they provide corrections for (GPS, Galileo, GLONASS...)

mischla, 2022-04-18:
Thanks for the quick answer, in this case all of them.
Link below in German valid for NRW: https://gppspro.saposnrw.de/service.php

stilsch, 2022-04-18:
Maybe a look at ardumower.de and it's Sunray (GPS-RTK) firmware might also help you https://github.com/Ardumower/Sunray

Bojkas1985, 2022-04-19:
Hi, wouldn't some form of triangulation using bluetooth iBeacon be possible?

fsommer1968, 2022-04-19:
> Hi, wouldn't some form of triangulation using bluetooth iBeacon be possible?

What is the accuracy you can achieve?
Everything more than a few (more than 10cm) over the complete backyard is just not acceptable!

GruNyv, 2022-04-19:
From what I could find on the Internet, the accuracy is at best one meter, which is not good enough to be used for control. Could maybe be used as a backup if you add a safety zone on 1 meter from the perimeter when you don't have RTK, but it will of course increase the already pretty expensive build

lightheaded, 2022-08-07:
Hi! Coming back to the initial question, would it be possible to add a "getting started" section to the readme? I have extensive software background and I tinker with some electronics, but I've never created my own PCB. Looking at the project, I have the same problem as @chicco83 - where to start?

vamatis, 2022-08-10:
Have a look at the wiki. Most of the needed infos are there...
If in doubt, just ask in the discord

lightheaded, 2022-08-10:
Thanks for pointing me that way! Here's the link for those still lost: https://wiki.openmower.de/index.php?title=Main_Page#How_to_Get_Started

rfvermut, 2023-02-13:
There is a better guide now: https://openmower.de/docs/getting-started/