ClemensElflein/OpenMower #5: WORX Landroid M WR140

TonyBrobston, 2022-04-11:
> Naturally I tried to dig deeper and actually found evidence that the mainboard is manufactured by some chinese company (SUMTEC Hardware).

This part of the readme got me curious to checkout the board in my WORX Landroid M WR140.

It looks like it's a completely different board; however, I figured since I took the pictures, I mine as well post them in case this mower can somehow be supported. If that is the case, I'm definitely interested in getting off Worx horrible software, it is quite unimpressive.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

unreality, 2022-04-14:
Depending on whether that ESP32 runs the whole thing or just provides wifi/bluetooth connectivity you could reflash it quite easily with custom firmware.

The ESP32 is reasonably powerful for an embedded soc.

LTKingChaos, 2022-04-16:
I would also like custom firmware for the Landroid. Factory firmware sucks bigtime.

hugokernel, 2022-04-20:
Thank you for the pictures @TonyBrobston 

I'm also very interested in an Opensource version of a Landroid (WR155 / L2000) in order to perpetuate the use of my robot and to bring additional features.

Seeing the electronic board and the fact that the board is embedded in resin will make it very difficult to repair.

The LandLord project (https://github.com/Damme/LandLord and https://hackaday.io/project/6717-project-landlord) is trying to produce an Opensource software for Worx robots (specifically the wg79X).

Damme, 2022-04-20:
Btw, the cpu is a LPC1768 under the goop (epoxy?)

RFBomb, 2022-05-06:
I have a 2021/22 model WR147 (with the radial dial for menu navigation), and I am also interested in this project. I also have the ACS unit installed, so support for that would be a must.

Michael-Bischof, 2022-05-07:
Also super interested in flashing custom firmware on the worx robot. I have the Landroid L (WR148E) and would also be into modding it. Can‘t get worse than it is right now software wise 😅

JochenKr, 2022-05-11:
Any chance that LandLord project and OpenMower will be combined?
OpenMower idea to use Rpi with good GPS is fantastic. This in combination with a simple modification of the Worx Landroid e.g. just connecting the Rpi to the original MCU to give the commands to the robot, would be a great simplification and super cost efficient

MBischof-gcx, 2022-05-11:
Already happening ;) See https://github.com/Damme/LandLord/issues/5

endoskop27, 2022-05-23:
Hi @TonyBrobston! Is there any Letters/Numbers printed on the big Controller on the Board (not the ESP32)?
On the pictures it is quite invisible

chopperpl, 2023-05-08:
Is this still alive? I would love to jump on the wagon, but it seems the shortage of RP boards and its price spike have killed the project.

mangeamg, 2023-05-24:
Why not use a cheap RPi clone, like orange?

scigany, 2023-07-12:
Hi,
does anyone have a diagram where I can identify the two elements marked with an arrow that are burnt in mine?
burned a hole in them,

[image]

Jmaxime89, 2023-07-29:
Hello ! I too would be sooo interested in supporting the landroid.

I keep an eye on the projet since a long time but the yard force is very hard to source in canada

pawster-hub, 2023-08-04:
Good morning,
I have a pcb in front of me with fried components. Can someone give the symbols of the elements t9, t10, t? (left of connector), d5, ic9? Hopefully I'll be able to do something more with it. Regards
[image]

shagi898, 2024-05-25:
Hi. someone found ic9
[image]

iLLiac4, 2024-06-09:
What is the purpose of this IC. The problem is that even when robot is turned off with the inserted battery the chip consumes power. I have checked it with thermal camera. This ic gets at around 30-40C. Also the coil gets a little warm. The problem is that it drains the battery even when not doing anything. Worx batteries has no overdischarge protection in them so it drains them fully.

[image]

Andy617237, 2025-03-24:
> Hi. someone found ic9 [image]

Hello, Someone find what is iC9 my is burned ;-(