ClemensElflein/OpenMower #6: Alternative to custom PCB

conradkirschner, 2022-04-13:
Is there a possibility to remove the custom PCB and use "standard components" e.g. pi cluster ?

This would enable more people in development.

fsommer1968, 2022-04-13:
You have to integrate with the remaining parts like left drive, right drive, cut engine. I´m not aware of a pi that can direct connect :-) Therefore at some point you always need an integration component, you name it "custom PCB". I applaud the effort building the PCB because I´m sure it costs a lot of money.

conradkirschner, 2022-04-13:
Would be cool to have the research material why a custom PCB was required and where it could be possibily replaced. 

I defenitly see the reason to use a custom PCB, it's just for enabling more devs if this step would be optional for production use ;)

r00li, 2022-04-13:
I mean these days where you can get a custom PCB made for like 5$ (a bit more for bigger ones, but still not that much) it's a no-brainer that a custom PCB will be used. I don't see why one would even want to go towards a different route.

ClemensElflein, 2022-04-13:
I agree, it's an obvious choice here. We need to connect hardware of some kind to interface with the robot and we need it to fit a predefined space. A custom PCB provides all that for very cheap.
I know that not everyone is willing to solder the SMD components, but that's a completely different issue. At the moment this is very much R&D so there's no way around it.

closed 2022-04-13 by ClemensElflein