ClemensElflein/OpenMower #8: US/North American source for compatible base hardware?

aderusha, 2022-04-14:
As noted in the README, the existing platform is the Yard Force Classic 500 which comes from a Chinese OEM [SUMEC Hardware and Tools](https://www.sumecpower.com/) which itself is a subsidiary of [SINOMACH](http://www.sinomach.com.cn/en/).  Their ["Brand" page](https://www.sumecpower.com/brand) lists G-Force, Yard Force, and PowerG.  PowerG looks to be battery powered hand tools, which leaves the G-Force and Yard Force side of things. 

## Yard Force USA

The Canadian and US links from the SUMEC page bring you to [YardForceUSA.com](https://www.yardforceusa.com/).  They have a [Robot Mower product page](https://www.yardforceusa.com/robot-lawn-mowers) with two products listed, the RVX4E and RVX4.  I can find no mention of these devices being made available for sale anywhere.  I've emailed the US distributor and will update if I get an answer back.  I eventually found [an online shop for Yard Force tools in the US](https://sumec.mybigcommerce.com/force-tools/) but no robot mowers are listed.

The [international page for Yard Force](https://www.yardforce-tools.com/PC_Web/International/HTML/Products.html) lists different models than the NA page, and that includes a model [SA500ECO](https://www.yardforce-tools.com/PC_Web/International/HTML/Products_parameter.html?id=598) which bears a passing resemblance to the Classic 500 (as well as being suspiciously similarly named).  [Amazon in the UK](https://www.amazon.co.uk/Yard-Force-Robomower-Lithium-Ion-SA500ECO/dp/B07B6WHCYQ) shows this model as no longer available and it appears to be an older 2018 device which they possibly are no longer making.

## G-Force

This line is harder to pin down.  [The product page](https://www.gforce-tools.com/en/RoboticMower-26.html) shows a handful of models, but I cannot find any listings for these mowers on sale anywhere.  [Product videos I've found are from 2015](https://www.youtube.com/watch?v=HbvwlFQ7TGM), so this might be an older line of devices.  

---

If anyone else has any ideas as how to obtain a suitable mower deck in the US, post here and we can all get to hackin'!

llama, 2022-04-22:
I am planning to order from [amazon.de](https://www.amazon.de/M%C3%A4hroboter-Classic-500-Kantenschneide-Funktion-Wochentags-Bedienfeld/dp/B08H1Y3KXZ?keywords=yard%2Bforce%2Bclassic%2B500&qid=1649430969&sprefix=yard%2Bforce%2Bclassic%2Caps%2C81&sr=8-3&linkCode=sl1&tag=selfomat-21&linkId=a2a7174df2bf2fb665403a4c9df5cbda&language=de_DE&ref_=nav_signin&th=1)  and use [mygermany.com](https://account.mygermany.com/web/content/new-register?canvassID=cdbe1208-010e-47b1-9c4d-8b1bb678f9e4) to forward the package.

The question I have: do all of the models on that Amazon.de page work with OpenMower? Or should I buy the "Classic 500" to exactly match what @ClemensElflein has done.

rjt, 2022-05-25:
$590 New up on eBay.  Buying thru PayPal may offer some protection even though [drahman7](https://www.ebay.com/usr/drahman7/) is a member of this project.  [https://www.ebay.com/itm/304477543176?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=E2AUex2jTNW&sssrc=2047675&ssuid=Yrw_Ogb9TDi&widget_ver=artemis&media=COPY](YARD FORCE robot lawn mower Classic 500, 500 m², 40% incline, works w OpenMower)  

[https://www.ebay.com/itm/304477543176?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=E2AUex2jTNW&sssrc=2047675&ssuid=Yrw_Ogb9TDi&widget_ver=artemis&media=COPY](https://www.ebay.com/itm/304477543176?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=E2AUex2jTNW&sssrc=2047675&ssuid=Yrw_Ogb9TDi&widget_ver=artemis&media=COPY)

darkmastermindz, 2023-04-14:
Any updates on this?

ffries, 2023-07-10:
If you visit Amazon in France, you should be able to find the product number (ASIN). I did the converse many times. The ASIN number is B09RFVFV1C. Then connect amazon.fr and try to buy it.

Here is a full URL:
https://www.amazon.fr/Yard-Force-Tondeuse-Classic-Bluetooth/dp/B09RFVFV1C/

Please note that I did not purchase it myself and I am still waiting.

symmatree, 2024-04-15:
I can't find the Yard Force 500 Classic for sale on any site now. I'd guess some other chassis would work but it's not at all clear how to validate that?

liquidbear99, 2024-08-03:
@symmatree I also am unable to see it sold anywhere 
on any site. (North America or Europe) The classic 500 may be discontinued.

mpcathey, 2025-06-26:
I'm in the same boat.  Has anyone else had better luck?