ClemensElflein/OpenMower #9: Add issues for development

conradkirschner, 2022-04-19:
Since there are some open tasks todo (e.g. build an app) I would be super happy about a roadmap/some "good starting point" tickets. 

Maybe a Roadmap is a good starting point, from there we could create some tickets/tasks and some kind of structure (tags)

ncx94, 2022-04-23:
Hi,

We are working on having the roadmap available on the wiki :)
Hopefully it will be online soon!

pedromorgan, 2022-05-04:
Its a "mediawiki install" > https://wiki.openmower.de/index.php?title=Main_Page

Need to discuss this.. and basically a markdown site would suffice in most cases...
- the backend and hosting of this is a huge burden into future.. imho...

gitlab pages and Ci is a free versionable and same caper, but a static website instead...#

pedromorgan, 2022-05-04:
Can help  with replacing wikimedia.. sooner the better..

closed 2023-06-28 by ClemensElflein