ClemensElflein/OpenMower #10: Avoid filesystem corruption

vguttmann, 2022-04-24:
I would very much like to build the mower and help with development, but for me to build it, it needs to be foolproof long-term, as I will be moving out soon, and I don't think my parents have any interest in reinstalling the operating system on their lawn mower.

Raspberry Pis have a history of corrupting the file system on their SD cards, so I'd like to know if there are any mitigations for this, and if not, advocate for their implementation.

Schaumi19, 2022-04-25:
As far as I know the OS is still Raspbian with ROS installed. The Problem with the SD-card is writing and in Raspbian there is a option to not write to SD. If that is selected it writes only to the RAM so if there is a power loss the system is set back to the point the option was selected. 
Hope it helps

noobydp, 2022-04-29:
I got bored and read up on this problem, so I'm no expert. 

[This article](https://hackaday.com/2022/03/09/raspberry-pi-and-the-story-of-sd-card-corruption/) suggests the problem has been resolved and may have been due to bad rpi hardware or cheap sd cards. 

I've used 3 or 4 pi's over the past few years starting with a Pi B, and I've never had an issue. 
I've always Used genuine samdisk cards, not sure if it helps. 

That said the overlay fs/ read only sd card mode is handled at the os level in raspbian so might be an option.

scottishjohn1982, 2022-04-29:
The Amazon warehouses in the us are full o of "legitimate" sd cards from
major brands that are actually produced there manufacturer but are part of
lots that failed qc and somehow (through illustrated supply chains) work
their way in to the Amazon warehouses and get mixed back into the
legitimate supply chain.

I have a stack of sd cards that I paid good money for that I only used for
projects that can run in read only mode.

There were issues with the older pis that could easily corrupt quality sd
cards, but those have been resolved.

Schaumi19, 2022-04-29:
It is true that you have less problems with genuine sdcards whatever brand, but it is also a fakt that the life time of an sd card is limited tue to the write cycles(up to 10k some sources say 100k). Thats why the problem is not the normal usage. But if there is something like a log file that gets written every second you are done afer a few days. Or if the file gets writen less the problem occers afer a longer period of time what could be worse.

FilipBE, 2022-04-30:
Some Raspberry Pi Compute Module 4 (CM4) boards come with a built-in eMMC flash chip for storing nonvolatile data. This chip replaces the need for a separate SD card, which can be useful if you want to avoid the use of SD cards.

JoeKae, 2022-04-30:
Just boot from a small ssd via usb. They last much longer than sdcards or emmc.

FadeFx, 2022-04-30:
i am not sure if adding an ssd inside the smal case of a mower.

JoeKae, 2022-04-30:
You can use small m2 type ssds.

vguttmann, 2022-04-30:
I did open this issue because I had a (as far as I can tell) genuine Samsung 64GB SD card go bad in a Pi that a personal project ran on.

I would agree that a CM4-based mower sounds like a good idea, given the onboard eMMC flash, and that it is intended with use in industrial settings in mind. Of course, that would mean sacrificing USB 3.0 connectivity, but as far as I can see, that's not really used here.
The CM4 DOES have a USB 2.0 interface, so no USB controller on the board would be necessary.

pedromorgan, 2022-05-04:
At end of day we want machine to be 100% reliable.. with ..a failover etc...(line replacable)

maybe an SD raid ? or ssd RAID ? 
as @JoeKae mentions, the m2 ssd's could be useful.. but expensive..

or a stpuid raid..
https://hackaday.com/2018/02/12/worlds-stupidest-solid-state-disk-drive-hack/

Schaumi19, 2022-05-05:
ssds last longer but have the same problem. So just dont write do the ssd/sd or only write every few days. I don't think it is a real problem, that it gets reset a few days if the battery dies.

wooter, 2022-06-21:
I've been using [Log2Ram](https://github.com/azlux/log2ram), and endurance type SD cards. None of them have failed yet, in other applications like RPi dash cams and power monitoring.

ClemensElflein, 2022-06-21:
We're working on having the software as docker images, so that we can use Raspbian as base system. This would allow us to use the overlayfs setting in raspi-config which should fix the sd corruption problem without much work.

plopaur, 2022-08-08:
I would recommend looking at Tinycore Linux for Rasperry Pi. This is a minimal ramdisk only distribution with no automatic writing to sdcard which i have used in the past. A new package for openmower has to be created.
Configuration changes can be written to sdcard instantly.

http://www.tinycorelinux.net/ports.html

docgalaxyblock, 2022-08-31:
@plopaur I could not find any information if ROS runs on TinyCore. If it does and it is stable that may be a way to do it

@vguttmann @pedromorgan I am planning to build my mower based on a CM4 w/ emmc and [this adapter](https://waveshare.com/wiki/CM4-IO-BASE-A) - It has a M.2 2230/42 slot too but I think a ssd is an overkill

It will take some time cause I accidentally ordered a CM4 w/o wifi so I am returning it and need to wait until my desired variant is available again.
Also I plan to take a look around the OS side of the project
*Stay tuned* 🤗