ClemensElflein/OpenMower #13: Cutting grass in different pathways

Frosty21, 2022-04-26:
More of a feature idea but when cutting grass you wouldn't want it to cut the grass the same way every time. So cutting it at 4 different angles (horizontal, vertical and diagonals) would be a good idea. That way you avoid ruts [ref](https://www.scotts.com/en-ca/library/spreaders-mowing-tools/how-mow-lawn-lawn-mowing-tips-tricks#:~:text=Each%20time%20you%20mow%2C%20do,forming%20ruts%20in%20the%20lawn.) [ref2](https://blog.davey.com/2019/08/should-i-alternate-directions-each-time-i-mow/)

dunno how you would implement this but if I have time later in the year I look into it from your code.

btw nice work @ClemensElflein . Maybe from this we can implement this as a snow remover (just an idea on top of this way out of left field...). would be cool if you could add another motor to horizontal attaching a snowblower scraper and the flat motor as a fan blower with a controlled vent that could store the snow in a certain area.

patrickdk77, 2022-04-29:
Is the gps accurate enough that it would ride in the same 1" wheel width track each mowing? Relative accuracy is pretty good, for position of a single mowing, but between days, or even hours, it should drift by a good foot or so, causing it to not rut.

staysayo, 2022-05-08:
Yes, it's very accurate because he's using RTK positioning that's accurate to 2 - 5cm (3/4" - 2").