ClemensElflein/OpenMower #39: Gardena Mower Compatibility

JasperMC, 2022-11-09:
Hi fellow robot mower owners,

I was wondering if anyone has tried converting a Gardena mower with this project.

Some Gardena mowers have a slot for a smart module, and I am wondering if that could be a way into controlling it without replacing hardware.

samborambo, 2023-04-03:
I have a Gardena Minimo 500. First thing I did was pull it apart to see how easy it would be to modify for OpenMower.

The motor controllers, charging system, STM32 microcontroller and bluetooth are on one conformally coated main board with the charging terminals mounted on it. Test pads and prog pins are exposed. I clean replacement would be to also replace the charging terminals with a 3D printed piece and some copper sheet.

All motors are brushless DC with integrated hall sensors, by the looks of it. Plenty of space inside around the main board. Nominal 18V system.

There's an RJ45 flat cable to another module at the rear, containing a USB-B connector that is accessible externally from a rubber plug. I haven't probed it yet. I'm guessing that the inductive sensor is also in this module.

That's pretty much all there is to the board. I'm just about finished setting up my perimeter wire. I'll probably start planning a mod and report back once it's working in normal operation first.

How well do DGPS sensors work under a thin plastic shell?

Sam.

patrickjsimons, 2023-04-03:
I've got two from the Sileno range, both of which have different boards and sensor arrangements. The main board in the 1600 older model gets really hot and the screen won't come on, so it looks like something on the board has short circuited internal, so it's time to take the OpenMower approach.

My first task is to modify one of the PCB designs in this repository to fit the OEM slot.  

Someone mentioned on Discord that this is quite easy to do.

mikechristiansen, 2023-04-03:
Hi Patrick,
I might be involved in the chat on discord. I have 2 dead Husqvarna Automower 105 and a 305 which share the same PCB. I want to re layout the pcb to make it fit in my mower but have never used pcb design software and have a few other commitments at the moment so not too much time to learn. I do wonder if your mower shares a similar pcb? I know Gardena, Husqvarna and maybe others share some tech. I'll try and draw up the footprint and some photos of my pcb. 
I do wonder if we can cloudfund someone to do the pcb layout work.
Thanks

patrickjsimons, 2023-04-03:
Hi Mike, When you say dead, is it battery related or do they both just not turn on?  I think after about three years these mowers have reach their life expectancy, whether it's motors, sensors or main board.  There is just no proper way too productively troubleshoot the main circuits, hence the reason I'm reluctant to just replace it with the OEM electronics.  I'm happy enough the way it chaotically approaches a lawn, as it's doing the job very well, so I'm more interested in being able to pinpoint issues and replace components. I'm also interested in moving away from the proprietary charging station as one of these has already been replaced under warranty.  I'll explore the option to resize the PCB and let you know how this pans out. In my case, it's probably more about stretching the real estate so it can be mounted, as opposed to reorganising the components.  Join the chat under Discord #open-mower-hardware

closed 2023-06-28 by ClemensElflein