ClemensElflein/OpenMower #95: Test for and fix overvoltage conditions

ClemensElflein, 2024-05-13:
when the mower is fully charged and the mowing motor breaks, the induced voltage spike might be too high.

- we need to verify that this is happening
- if it is happening, we need to protect against it

quelqundev, 2024-05-13:
Hi ! I would recommend to never charge the battery at full charge to be able to absord current rejection. Is it possible to configure a max charge voltage ?

Furthermore i belive it is better to avoid little cycle charge at high % (i mean it "costs" more cycle charge to do 20%=>50% than 70%=>100%). Maybe adding a max charge at 80% would allow to improve the battery longevity ?

I'm trying to help, thanks for your great job guys !! 👍

ClemensElflein, 2024-05-13:
you can set the max charge voltage to whatever you like, what do you mean it "costs" more to charge the last 30% than other 30%?

anyways the original issue is still relevant for people who decide to charge the battery to 100%

quelqundev, 2024-05-13:
I recommend you to read this good article : https://accubattery.zendesk.com/hc/en-us/articles/210224725-Charging-research-and-methodology

The more important principle about Lithium battery is to stop charging before the "constant phase".
Thus i consider it is to you to find a balance between charging at full capacith but "degrading" battery capacity faster...or charging at 80% but preserving battery capacity for longer time. 🔋

It's like a bonus that allow you to solve the hypothetic case of current rejection. 😎