ClemensElflein/OpenMower #108: parkside robot lawnmower

evlas, 2024-08-27:
Hi,
considering the low cost of the 2 parkside robot lawnmowers (PMRA 20-Li and PMRDA 20-Li), is it possible to make them compatible with the Openmower Project?

ryanm101, 2024-09-30:
Mine has stopped working and i was thinking about this

quelqundev, 2024-09-30:
Hi, you can check the discord there is some people working on it.

Nekraus created the following repo : https://github.com/Nekraus/PARKSIDE_PMRA

fercod, 2025-04-17:
I saw it at Lidl and immediately came here to comment on it.

ValAnd65, 2025-05-14:
arrivata oggi.
Vista la lentezza ed i disagi non credo vi terrò in mente per il futuro

ClemensElflein, 2026-09-03:
Parkside mower is compatible with the universal hardware. Please check the repos or ask on Discord for more details

closed 2026-09-03 by ClemensElflein