ClemensElflein/OpenMower #117: Possibility LIDAR sensor?

tirithen, 2025-06-08:
This is a super cool project. Might be interested of the version 2. But still I'm wondering about obstacle avoidance to deviate from the path if there are toys or people on the lawn during the run. There are LIDAR sensors for around €90 or so  https://www.amazon.se/-/en/youyeetoo-FHL-LD19-Lidar-Sensor-Resistant/dp/B0B1QCV4XR/262-5451151-7989726?psc=1 .

The description do mention ROS/ROS2 and raspberry pi. I wonder if something like that might be an option. No clue if it is weather resistant over time or not.

Either way, keep up your cool work, and keen to see more on the version 2 announcement.

barateau, 2025-08-27:
Hey hey, I have followed your wonderful project quite some time and have been dreaming of you mod for my yardforce OEM model. Anyway i did watch an interesting talk with greatscott and he highlighted some new IC thats high frequency radar.  Basically 57-64GHz  radar sensor with great visibility, this might be the actual game changer compared to the bumpers or even lidar that I feel even fail in a home environment.

Please have a look and say what you think about the functionality, as I sould say it could perhaps be a really good upgrade for version V2.5

https://www.ti.com/product/IWRL6432

[image]

barateau, 2025-08-27:
Did add a Q to greatscott, if he can give some insight on how to implement or leverage. Not only on the PCB but it would need a logic or trained LLM model to interpret the radar senor.

https://youtube.com/watch?v=D86an4uN7FE&lc=UgxEqrR4quBw1_4g5lF4AaABAg&feature=shared